iarsystems/iar-vsc-build #4: Feature : Compile output on VS code output window

xNiux, 2022-06-16:
It could be very usefull to have the compilation output on VSCode Output tabs instaed of Terminal one.
==> Output tabs could be use color to highlight Error, Warning and others ...

For example, I combined this 2 extensions : 
- [https://marketplace.visualstudio.com/items?itemName=zip696.iar-tool](https://marketplace.visualstudio.com/items?itemName=zip696.iar-tool)
- [https://marketplace.visualstudio.com/items?itemName=IBM.output-colorizer](https://marketplace.visualstudio.com/items?itemName=IBM.output-colorizer)

[image]

And the error is highlight in RED, and Warning in Orange. It is really cool :)

HampusAdolfsson, 2022-06-17:
I really like the idea of having colorized output for building.

Since this extension uses `Task`s for compilation/build jobs, I'm not sure if there is a "good" way to have it output to the Output tab instead of the Terminal. I think a better solution is to inject terminal escape codes to stylize the build output in the Terminal tab. I'll experiment a bit with scanning the build output and adding some color to it.

xNiux, 2022-06-17:
> This is just a quick prototype, but an example of what it could look like:
> [image: image]
> <https://user-images.githubusercontent.com/15944368/174312504-149e9417-1343-46f2-bbe5-43bab13a2228.png>

Great! It's sounds vert well!

HampusAdolfsson, 2022-07-06:
This is now released.
There might be room for improvement in the future, but for now we highlight:
* Files
* Errors/Warnings
* Error/warning codes

closed 2022-07-06 by HampusAdolfsson