iarsystems/iar-vsc-build #5: Support for .custom_argvars files.

nxj, 2022-06-21:
Last thing missing for me to be able to use this extension is support for .custom_argvars files. As a workaround i can just add the "-varfile C:\path\to\project.custom_argvars" to the extraBuildArguments in the workspace settings. But this only works with an absolute path and because variables are not allowed in the settings, it is not very portable. Would be great if some general support for these files could be added!

HampusAdolfsson, 2022-06-21:
Until we implement proper support for this, would it help if we allowed build arguments to be relative to the workspace?
Then, if the open folder is `C:\path\to`, you could simply specify `"-varfile project.custom_argvars"` to mean `C:\path\to\project.custom_argvars`. It would be a simple change to make and would make the setting more portable.

nxj, 2022-06-22:
That sounds great and would help!

ycsin, 2022-07-15:
I wonder if this is in the 1.10.4 already?

[image]

I'm getting this error

```
     IAR Command Line Build Utility V8.4.8.6680
     Copyright 2002-2020 IAR Systems AB.

 Usage:
 iarbuild <projectfile> [-clean | -build | -make | -cstat_analyze | -cstat_clean]
 <config> [-log errors|warnings|info|all] [-parallel <number>] [-varfile <argvarfile>]

 ERROR, Unknown command: "-varfile Project.custom_argvars" is not a valid command

WARN: Builder closed without reaching linking stage.
Source configuration did not complete: Process exited with code: 31
```

ycsin, 2022-07-15:
Fixed, the arg needs to be on the next item instead of inline
[image]

HampusAdolfsson, 2022-12-07:
Custom Argvars support has now been added, and you should no longer add extra build arguments for it. They are added automatically when you select a .custom_argvars file in the extension configuration view.

closed 2022-12-07 by HampusAdolfsson