iarsystems/iar-vsc-build #8: Buildable project but IntelliSense detects multiple errors depending on order of includes

mulleandreas, 2022-06-23:
I have multiple projects that compile flawlessly, but IntelliSense detects multiple issues. The error messages point in the direction that IntelliSense cannot find the definitions. If I change the order of the includes, IntelliSense can resolve them. I broke one of the projects down to a minimum sample and checked the includes and I'm quite sure that there are no issues with them. 

Also, if I re-add the previously used c_cpp_properties.json file to the project and remove the line `"C_Cpp.default.configurationProvider": "iarsystems.iar-build"` from the settings.json file, IntelliSense works regardless of the order of the  includes and detects no errors. 
[c_cpp_properties.json.txt](https://github.com/IARSystems/iar-vsc-build/files/8967751/c_cpp_properties.json.txt)

This issue might be related to [Incorrect intellisense configuration #6](https://github.com/IARSystems/iar-vsc-build/issues/6).

I would also be happy to provide the sample project privately.

IAR Build Log

```
[23.6.2022, 14:58:29][Debug] Activating extension
[23.6.2022, 14:58:29][Debug] Collected 5 toolchains
[23.6.2022, 14:58:29][Debug] Toolchain: selected 'Embedded Workbench 9.1 (Arm)' (index 3)
[23.6.2022, 14:58:29][Debug] Loading thrift workbench 'Embedded Workbench 9.1 (Arm)'...
[23.6.2022, 14:58:29][Debug] 4 new toolchain(s) added
[23.6.2022, 14:58:29][Debug] Found 3 project(s) in the workspace
[23.6.2022, 14:58:29][Debug] Project: selected 'bootloader-ra6m5' (index 0)
[23.6.2022, 14:58:29][Debug] Configuration: selected 'Debug' (index 0)
[23.6.2022, 14:58:29][Debug] Generating intellisense config for 'bootloader-ra6m5':'Debug'...
[23.6.2022, 14:58:29][Debug] Loaded thrift workbench 'Embedded Workbench 9.1 (Arm)'
[23.6.2022, 14:58:29][Debug] Loading project 'bootloader-ra6m5' using thrift...
[23.6.2022, 14:58:29][Debug] Loading project context for 'bootloader-ra6m5'
[23.6.2022, 14:58:29][Debug] Loading project 'bootloader-ra6m5' using thrift...
[23.6.2022, 14:58:30][Debug] Intellisense config changed. Notifying cpptools.
[23.6.2022, 14:58:30][Debug] Providing intellisense configuration(s) for: c:\Projekte\test\src\main.cpp
```

IAR Config Generator Log
```
Generating compilation database...

     IAR Command Line Build Utility V9.1.1.9611
     Copyright 2002-2022 IAR Systems AB.

bootloader-ra6m5 - Debug
Build description written to C:\Users\102-anmu\AppData\Local\Temp\iar-jsondbabcda958c3e05e627317c4b1d245b754.json

Done!
Generating intellisense information for c:\Projekte\test\src\main.cpp
```

C/C++ Extension Log

[c_cpp_extension_log.log](https://github.com/IARSystems/iar-vsc-build/files/8967638/c_cpp_extension_log.log)

HampusAdolfsson, 2022-06-27:
Hi,

This is probably a separate bug from #6. There is nothing that sticks out to me in the logs you provided, so if you could provide the project that would be very helpful. You can email it to me at:
`hampus.adolfsson@iar.com`

HampusAdolfsson, 2022-07-05:
This seems to be a problem with the parsing or caching of symbols, rather than with the configuration. I've reported the problem to `cpptools`:
https://github.com/microsoft/vscode-cpptools/issues/9540

JackHerRrer, 2022-07-28:
I don't know if this will help, but i had similar issues because of the now deprecated "IAR For Visual Studio Code" package by pluyckx. Simply disabling it was not enough.
To solve the issue, I disabled both extensions and deleted every configuration related to both extensions in all the config files of the project (c_cpp_properties.json, iar-vsc.json, settings.json). Then I enabled only the official IAR extension. Only then it started to work properly

omisys, 2023-05-16:
This is not an issue with cpptools. The issue is that only the folders that are set in either the additional include directories, Preinclude files or defines are being sent to the cpptools plugin as browse path. If your source(c) files are not in the same directory as the header files they will not be indexed. The source(c) files that are added to the IAR project should also be added to the browse path. If your project is setup like this:

proj/include
proj/src

And the proj/include folder is added to the include directories in the preprocessor, only the include files will be indexed. You can work around this problem by added the src folder to the include directories as well so it will be sent to cpptools. But I think this should be fixed in this addon. 

The C files that are used in the project are known to the EWP file so it should be possible to add them to browse path as well. I don't have the TS knowledge to make a pull request but I hope @HampusAdolfsson can pick this up? 

`    public provideBrowseInfo(): IntellisenseInfo {
        const config = this.workspaceIntellisenseInfo?.getBrowseInfo();
        return {
            defines: config?.defines ?? [],
            includes: config?.includes ?? [],
            preincludes: config?.preincludes ?? [],
        };
    }`

HampusAdolfsson, 2023-06-01:
As I wrote in #17, @omisys solution fixes that issue. However, this is not the same issue and it is unfortunately not fixed by adding source folders to the browse path (in fact the example I sent to the cpptools developers already has the correct browse path).

That means that this issue is still most likely a problem in cpptools.

kellpete, 2025-07-08:
It looks like the define `"__IAR_SYSTEMS_ICC__ =9"` is causing the trouble in intellisense. 
This define is used in libcpp to do things in a compiler-specific way. I guess the intellisense parser can not correctly handle these IAR compiler specifics.
When I take the config from the C/C++ console, copy that to my c_cpp_properties.json, and then comment out the define for `__IAR_SYSTEMS_ICC__`, intellisense works fine again.

HampusAdolfsson, 2025-07-09:
@kellpete Yes, there are unfortunately certain IAR compiler features that the intellisense parser does not handle well.

What specific parts of libcpp are you having issues with? We can have a look at them to see if there's anything we can do with the preprocessor to work around the problems (besides undefining `__IAR_SYSTEMS_ICC__`).

kellpete, 2025-07-09:
@HampusAdolfsson Thanks for the quick reaction! Unfortunately, I do not know which part of libcpp (or even DLIB?) causes the issues. Intellisense has trouble finding symbols (namespaces, classes) from our own codebase, but we do not use `__IAR_SYSTEMS_ICC__` at all. There are hundreds of #ifdef `__IAR_SYSTEMS_ICC__` in libcpp, and I do not know where to start searching. 

I will discuss in the team here, maybe someone has an idea how to narrow it down.

In meantime, if you can hint what constructs could cause such problems, that would be very helpful to know. You confirmed that intellisense has issues with certain IAR compiler features, but which ones? E.g. should I look for calls to __builtin functions? Also, is there a way to get intellisense parser debug output to see where it is failing?

kellpete, 2025-07-10:
Good news: I was able to find the offending header by putting `#undef __IAR_SYSTEMS_ICC__` into the headers until I found it.

The problem is caused in `\arm\inc\libcpp\__type_traits\common_reference.h`, which is included via the `array` header. The code in this file is wrapped in
`#if _LIBCPP_STD_VER > 17 || defined(__IAR_SYSTEMS_ICC__)`
The problematic lines are those that start with
`requires requires [...]`
Intellisense shows following error for these lines:
`argument list for variable template "std::requires" is missing`

**Fixes**
I found the following solutions

- Add parantheses: Copilot suggest to add parantheses like this: `requires (requires [...])`, which indeed solves the original issue in our code, but still shows an error for the requires: "this declaration has no storage class or type specifier
- An alternate solution I found is to change `cppStandard` in c_cpp_properties.json from c++17 to c++20

But I hope someone more knowledgeable with C++20 and `requires` clause can handle it from here. Let me know if I can assist with any more information!