iarsystems/iar-vsc-build #9: ERROR, Unknown argument variable

lkj0136, 2022-06-24:
error message 
==============================
 ERROR, Unknown argument variable: Failed to expand argument variable "$DATE$"
Source configuration did not complete: Process exited with code: 3221226505

my project file (ewp)
==============================
<?xml version="1.0" encoding="UTF-8"?>
<project>
    <fileVersion>3</fileVersion>
    <configuration>
        <toolchain>
            <name>ARM</name>
        </toolchain>
        <debug>0</debug>
        <settings>
            <name>OBJCOPY</name>
            <archiveVersion>0</archiveVersion>
            <data>
                <option>
                    <name>OOCOutputFile</name>
                    <state>$DATE$_$PROJ_FNAME$_$CONFIG_NAME$.bin</state>
                </option>
            </data>
        </settings>
    </configuration>
</project>

micaelbo, 2022-06-27:
Could you verify that this .ewp file works with EWARM and/or iarbuild?

micaelbo, 2022-06-29:
It is now verified that this is a problem with the command line utility iarbuild.exe which is part of the EW/BX product. The bug is listed as IAR issue [IDE-6189] "The $DATE$ argvar does not expand from IarBuild" to be solved in a coming release of IAR Embedded Workbench and IAR Build Tools.

closed 2022-07-01 by HampusAdolfsson