iarsystems/iar-vsc-build #15: Export selected project as variable so it is usable in tasks.

nxj, 2022-07-06:
Similar to the current project configuration it would be great if one could use the current project as variable in the tasks.
For example using ${command:iar-config.project} to access the hex file depending on the currently selected project:

```json
{
	"label": "My Custom Task",
	"type": "process",
	"command": "test.exe",
	"args": [
		" --hex_file ${workspaceFolder}\\path\\to\\iar\\${command:iar-config.project}\\${command:iar-config.project-configuration}\\Exe\\out.hex",
	]
}
```

HampusAdolfsson, 2022-07-08:
Do you mean just the name of the project? Or the path to the project directory?

There is `${command:iar-config.project-file}`, which gives you the full path of the .ewp file. You could do something like this:
```json
" --hex_file ${command:iar-config.project-file}\\..\\${command:iar-config.project-configuration}\\Exe\\out.hex",
```

nxj, 2022-07-08:
I meant the name of the project. Maybe my example was not very good. For my project, the name of the out file is also the project name so i would like to use it there too. E.g.

```json
" --hex_file ${workspaceFolder}\\path\\to\\iar\\${command:iar-config.project}\\${command:iar-config.project-configuration}\\Exe\\${command:iar-config.project}.hex"
```

HampusAdolfsson, 2022-10-07:
Implemented in 1.10.5

closed 2022-10-07 by HampusAdolfsson