iarsystems/iar-vsc-build #16: Cannot click on path in build output with space

spoorcc, 2022-07-18:
When a file path has a space in it, it cannot be clicked since both parts are see as separate paths:

`'c:\Projects\SomeProject\Some Folder\SomeFile.cpp'` will be linked as 2 separate paths:

* `c:\Projects\SomeProject\Some` 
* `Folder\SomeFile.cpp`

mario-pi, 2022-07-19:
Thanks for reporting this. Does this happen in the colorized build output introduced in https://github.com/IARSystems/iar-vsc-build/issues/4 ?

spoorcc, 2022-07-26:
Sorry, I was wrong 🙈, it only happens for link of the `.ewp` project file (and the path to IarBuild.exe) when starting the task.
I'm using version v1.10.4 of the plugin.

I tried clicking on the `.ewp` and the link was broken into parts:
```
Executing task: iar: Iar Build 

> 'C:\Program Files (x86)\IAR Systems\EWARM 8.50.5/common/bin/IarBuild.exe' 'c:\Projects\Customer\app\Core\Project Files\IPROG-85IC1-T1-APP.ewp' '-make' 'Release' '-log' 'info' '-parallel' '4'

     IAR Command Line Build Utility V8.4.8.6680
     Copyright 2002-2020 IAR Systems AB.
```

* `'c:\Projects\Customer\app\Core\Project` 
* `Files\IPROG-85IC1-T1-APP.ewp'`

mario-pi, 2022-08-02:
Ok thanks. 
Sounds like this might be related to VSCode issue https://github.com/microsoft/vscode/issues/97941 .
We have unfortunately limited capacity during the summer break so it might take some time to determine if it is actually caused by that.

HampusAdolfsson, 2022-08-17:
This is indeed caused by the VS Code issue linked above. 

As a workaround, we can add our own link provider for the .ewp file path that supports spaces (we already do this for other parts of the build output).

HampusAdolfsson, 2022-10-07:
Fixed in 1.10.5

closed 2022-10-07 by HampusAdolfsson