iarsystems/iar-vsc-build #17: Project is not fully indexed by IntelliSense until target .c file is opened

LonginKandinsky, 2022-07-21:
I´m often finding functions what seem not be fully indexed by IntelliSense and it´s just jump to prototype declaration in header file. It is typically for case when header and source file names are not exactly same - for example xxx_common.h / yyy.c etc

For this case the "Go To Definition" function is not functional until the .c file with definition is opened. Then the found declaration-definition linkage is somehow cached and it works until VSCode IAR project is closed.

Is this problem related to cooperation between IAR extensions and IntelliSense or just IntelliSense itself?

HampusAdolfsson, 2022-08-17:
This might be related to #8

heppson, 2022-10-13:
I have the same issue. Intellisense works fine until I install the IAR Build plugin, then it won't find the definitions anymore. I have the header files in a separate directory, included from IAR Project settings > C/C++ Comiler > Preprocessor.

Here is a minimal example project: [iar-build-extension-issue.zip](https://github.com/IARSystems/iar-vsc-build/files/9776757/iar-build-extension-issue.zip)

To reproduce:
1. Open workspace in VS Code
2. Open `project/inc/example.h`
3. Right-click `example` and "Go to definition". Nothing happens. (Did it work? Try restarting VS Code and try again.)

To "fix" the problem, disable IAR Build:
1. Remove `"configurationProvider": "iarsystems.iar-build"`from `.vscode/c_cpp_properties.json`
2. Close the example.c tab
3. Restart VS Code
4. Open `example.h` again and go to definiton. It finds the definition in `example.c` as expected.

stdockd, 2023-03-11:
Having the same issue, IntelliSense cannot resolve definitions and only jumps to declarations.

IAR Config Generator outputs "Generating intellisense information for file_name.c from project project_name" only when I open it said file in code editor.

HampusAdolfsson, 2023-03-13:
The C/C++ extension is responsible for indexing projects and provides most language features. This extensions doesn't do much in that regard other than respond with compilation parameters when they are requested by the C/C++ extension.

Some have reported bugs in the caching done by the C/C++ extension which can cause the problems you are experiencing. It may be that something in the way this extension reports compilation parameters triggers a bug in the C/C++ extension, but we haven't been able to find what that would be. A workaround that seems to work for some is setting `C_Cpp.intelliSenseCacheSize` to `0` in VS Code to disable the cache.

omisys, 2023-05-16:
Please check my comment on #8 @HampusAdolfsson

HampusAdolfsson, 2023-06-01:
@omisys I've confirmed that adding the source folders to `browse.path` solves this issue. We'll release a fix in the coming few weeks. Good work!

closed 2023-06-26 by IAR-build-bot