iarsystems/iar-vsc-build #19: Thrift workbench crashing

JackHerRrer, 2022-07-28:
Hello, I am trying to use an IAR project of my company with vscode, but I encounter difficulties. The IAR project view is unavailable.
The logs seems to indicate that the wrong workbench is selected, yet, when i open the project directly from this IAR workbench (The version from  C:\Program Files (x86)\IAR Systems\EW_ARM 9.20) everything is working fine. 

Do you now what the error code 3221225477 is ? Could it be related to the use of argvars ? 

IAR build
> [28/07/2022, 09:16:59][Debug] Activating extension
  [28/07/2022, 09:17:00][Debug] Collected 5 toolchains
  [28/07/2022, 09:17:00][Debug] Toolchain: selected 'EW_ARM 9.20 (Arm)' (index 0)
  [28/07/2022, 09:17:00][Debug] Loading thrift workbench 'EW_ARM 9.20 (Arm)'...
  [28/07/2022, 09:17:01][Debug] 2 new toolchain(s) added
  [28/07/2022, 09:17:01][Debug] Found 10 project(s) in the workspace
  [28/07/2022, 09:17:02][Debug] Project: selected 'Project1' (index 4)
  [28/07/2022, 09:17:02][Debug] Configuration: selected 'Debug' (index 0)
  [28/07/2022, 09:17:02][Debug] Generating intellisense config for 'Project1':'Debug'...
  [28/07/2022, 09:17:04][Debug] Loaded thrift workbench 'EW_ARM 9.20 (Arm)'
  [28/07/2022, 09:17:04][Debug] Loading project 'Project1' using thrift...
  [28/07/2022, 09:17:04][Debug] Loading project context for 'Project1'
  [28/07/2022, 09:17:04][Debug] Loading project 'Project1' using thrift...
  [28/07/2022, 09:17:12][Error] Thrift workbench 'EW_ARM 9.20 (Arm)' crashed (code 3221225477)
  [28/07/2022, 09:17:12][Debug] Loaded thrift workbench 'undefined'
  [28/07/2022, 09:17:12][Debug] Not loading project 'Project1' using thrift, no appropriate workbench selected...
  [28/07/2022, 09:17:12][Debug] Loaded extended project 'undefined'
  [28/07/2022, 09:17:16][Debug] Intellisense config changed. Notifying cpptools.

IarServiceManager 
>   IAR Service Launcher V9.0.11.9006
     Copyright 2018-2022 IAR Systems AB.
      [9164] 
      Service registry running on local socket on port: 60184
      Also available in serialized form in CSpyServer2-ServiceRegistry.txt
      [9164] Starting service manager service...
      [9164] Service: com.iar.ProjectManager
      [9164] Loading native library C:\Program Files (x86)\IAR Systems\EW_ARM 9.20\common\bin\ProjectManagerHandler
      [9164] Loaded module: 00007FFCFD670000
      [9164] Calling entry point to start service.
      [9164] Location: 
      [9164] Service started: com.iar.ProjectManager
      [9164] Entering main loop...
      Registry Exited (3221225477)

VasaMM, 2022-08-01:
I have same issue. I also use argvars (add by extra build argument - https://github.com/IARSystems/iar-vsc-build/issues/5#issuecomment-1185301113). If I remove usage of variables from _ewp_ file all works perfectly.

mario-pi, 2022-08-02:
Thanks for reporting this issue. The error code you are getting is most likely caused by the IarServiceLauncher process crashing.
Having a minimal ewp file + arg var causing the issue would be most helpful - is this caused by a specific arg var being used?

VasaMM, 2022-08-03:
I think, that any arg could cause this problem. I created blank project, it contains only very simple main and one custom argvar.
[test.zip](https://github.com/IARSystems/iar-vsc-build/files/9251967/test.zip)

JackHerRrer, 2022-08-04:
I tried the test project of @VasaMM. The same problem occured.

HampusAdolfsson, 2022-08-17:
The workaround by adding extra build arguments that @VasaMM mentioned only fixes building and intellisense, not the files view. As you have figured out, the workbench will crash when loading any project that uses an argvar file.

I'll consider this a bug; even if we don't support argvar files yet, the workbench should not crash if you try to use them.

HampusAdolfsson, 2022-12-07:
Custom argvars support has been added in the latest release. When you open your project with EWARM 9.20, you should get a warning instead of a crash. Please confirm whether this fixes your problem.

To work around the crash, the .custom_argvars file is not loaded into the "thrift workbench" that we use for the files view. This means that if you have argvars in the paths of your project files, they may not be expanded correctly.

JackHerRrer, 2023-01-26:
With the new version of the extension and IAR v9.32, I no longer have issues.
FYI, I had issues with argvars when using the new version of the extension and IAR v9.20. Upgrading to v9.32 solved it all.

closed 2023-01-26 by JackHerRrer