iarsystems/iar-vsc-build #21: Absolute paths used for include directories

allanwglen, 2022-08-16:
It appears that the extension converts "$PROJ_DIR$" paths from the .ewp file include directories to absolute paths before providing to 
 the Microsoft C/C++ extension configuration file, visible in file ".vscode\c_cpp_properties.json" under section "includePath".

It would make sense to translate IAR token "$PROJ_DIR$" paths to the equivalent "${workspaceFolder}" token to prevent invalid paths if the c_cpp_properties.json file is checked into source control and then checked out and compiled by another user. However, this only affects the red squiggles VS Code produces, as IAR compilation will still work fine using the .ewp file.

HampusAdolfsson, 2022-08-17:
Hi, 

Perhaps I am misunderstanding you in some way, but we send include directories to the Microsoft C/C++ extension via their typescript API. We do not use or modify the c_cpp_properties.json file in any way (usually you do not even need to have a c_cpp_properties.json).

The (deprecated) "IAR for Visual Studio Code" extension used to write directly to the c_cpp_properties.json file, but this extension does not.

allanwglen, 2022-09-08:
I believe that I have an old c_cpp_properties.json file hanging about from the depreciated "IAR for Visual Studio Code" that caused me to think it had come from this extension, you are correct.

closed 2022-09-08 by allanwglen