iarsystems/iar-vsc-build #24: Add specific color for #pragma message

xNiux, 2022-08-22:
Hi, to have more visual message during compilation, it could be very useful to have a dedicated color (ex: Yellow) when pragma message is displayed during compilation : 

[image]

[image]

xNiux, 2022-08-22:
To simplify, maybe the using keyword like "INFO" in the pragma could be a good solution : 

`#pragma message ("INFO: This is a debug version, do not use for production ....") ;`

HampusAdolfsson, 2022-08-26:
We can definitely add some color to highlight any line that starts with "INFO" (or "#").

Highlighting *any* pragma messages isn't really possible just from the VS Code extension, since we don't have any information about what caused each line to be printed. It would have to be added in the compiler itself (which is a bit more complicated).

HampusAdolfsson, 2022-12-07:
Sorry for the late feedback on this. We've decided not to implement this feature. This is something that should be implemented in the compiler rather than adding a half-measure solution to the VS Code extension.

closed 2022-12-07 by HampusAdolfsson