iarsystems/iar-vsc-build #27: Fileview says it requires 9.20.4 - but it is installed

TLorenzPH, 2022-08-26:
It says I require a newer IAR - but all version are installed

[image]

[image]

HampusAdolfsson, 2022-08-29:
The view is only supported when you have EWARM 9.20.4 or later selected as the active installation in VS Code. If the selected installation (i.e. 'Embedded Workbench 9.0_2') is older, the file view will be disabled (even if you have a newer EWARM version on your computer).

As a side note:
If you want to use the files view with EWARM 9.20.1, there is a way to enable it.
Go to your EWARM 9.20.4 installation, copy the file `common/bin/projectmanager.json`, and paste it into the same folder in your EWARM 9.20.1 installation. After reopening VS Code, the file view should be available with EWARM 9.20.1 (note that we don't formally support this, so there may be bugs).

TLorenzPH, 2022-09-06:
Hi.
On my PC the 9.20.4 is installed in folder 9.0_3....

So how can I solve this?

[image]

TLorenzPH, 2022-09-06:
I seems the project needs to be stored with 9.20.4 IAR- so the error meassage is missleading to some point.

HampusAdolfsson, 2022-09-07:
Hi,
In order for the files view to work, you need to select your EWARM 9.20.4 in the extension. In the dropdown in your first screenshot where it says "IAR Embedded Workbench or Build Tools installation, you would have to select the 9.0_3 folder. Note that if you then add or remove files to the project from VS Code, it will upgrade the project to 9.20.4, so you might not be able to open it in 9.20.1 anymore.

I agree that the error message could be clearer.

kzoltaan, 2023-01-26:
I have same issue. I'm using 9.20.2 on my laptop, and on my desktop. Same project on both computer. The laptop is working well, but my desktop computer message is "The IAR project view requires IAR Embedded Workbench for Arm 9.20.4 or later.". Its worked well. I dont know what is going wrong. Do u have any idea?

HampusAdolfsson, 2023-01-27:
The minimum required version for the files view is 9.20.4, so 9.20.2 is not expected to support it. However, as I mentioned above you can force enable the files view for 9.20.2 by creating the file `common\bin\projectmanager.json` in the installation folder (e.g. `C:\Program Files\IAR Systems\Embedded Workbench 9.0\common\bin\projectmanager.json`) with the following contents:
```json
{
    "services":[
      {
        "name":"com.iar.ProjectManager",
        "libraryName":"ProjectManagerHandler", 
        "spawnNewProcess":false,
        "startupEntryPoint":"StartProjectManager",
        "shutdownEntryPoint":"StopProjectManager",
        "registerInLauncher":false
      }
    ]
}
```
Afterwards, if you restart VS Code you should be able to use the files view.

kzoltaan, 2023-01-27:
Yes. This is the difference between my laptop and my desktop. My laptop has this file! Thanks!

closed 2023-06-26 by HampusAdolfsson