iarsystems/iar-vsc-build #28: MSP430 compiler is not found for generating intellisense configuration

LukasP, 2022-08-29:
When selecting an MSP430 toolchain the compiler executable is not found. 

The extension looks for the folder by toolchinaId, which is `MSP430`, However the folder for the MSP430 toolchain is simply called `430` (eg: ```C:\Program Files (x86)\IAR Systems\Embedded Workbench 8.5\430\bin```) resulting in the executable not being found and no configuration being provided to intellisense. 

IAR config generator log: 

```

     IAR Command Line Build Utility V8.5.1.7337/W32
     Copyright 2002-2021 IAR Systems AB.

Building configuration: MSP430F5257_clocktest - Debug
Updating build tree...
e:\MSP430F5257_clocktest\Debug\Obj\main.r43
e:\MSP430F5257_clocktest\Debug\Exe\MSP430F5257_clocktest.d43
e:\MSP430F5257_clocktest\Debug\List\MSP430F5257_clocktest.map

3  file(s) deleted.
Updating build tree...

main.cpp 
icc430.exe "e:\MSP430F5257_clocktest\main.cpp" "-o" "e:\MSP430F5257_clocktest\Debug\Obj" "--no_cse" "--no_unroll" "--no_inline" "--no_code_motion" "--no_tbaa" "--debug" "-D__MSP430F5257__" "-e" "--double=32" "--dlib_config" "c:\Program Files (x86)\IAR Systems\Embedded Workbench 8.5\430\lib\dlib\dl430xlsfn.h" "--core=430X" "--data_model=small" "-Ol" "--multiplier=32" "--eec++" "--hw_workaround=CPU40" "--code_model=large"

Done!
Generating intellisense information for e:\MSP430F5257_clocktest\main.cpp
WARN: Compiler gave error: Error: spawn c:\Program Files (x86)\IAR Systems\Embedded Workbench 8.5\msp430\bin\icc430.exe ENOENT

```
IAR build log: 
```
[29/08/2022, 19:36:24][Debug] Activating extension
[29/08/2022, 19:36:24][Error] Failed to fetch toolchains from registry: Error: Process exited with code: 1
[29/08/2022, 19:36:24][Debug] Collected 1 toolchains
[29/08/2022, 19:36:24][Debug] Toolchain: selected 'undefined' (index undefined)
[29/08/2022, 19:36:24][Debug] 1 new toolchain(s) added
[29/08/2022, 19:36:24][Debug] Loaded thrift workbench 'undefined'
[29/08/2022, 19:36:24][Debug] Loaded extended project 'undefined'
[29/08/2022, 19:36:25][Debug] Found 1 project(s) in the workspace
[29/08/2022, 19:36:25][Debug] Project: selected 'MSP430F5257_clocktest' (index 0)
[29/08/2022, 19:36:25][Debug] Configuration: selected 'Debug' (index 0)
[29/08/2022, 19:36:25][Debug] Not loading project 'MSP430F5257_clocktest' using thrift, no appropriate workbench selected...
[29/08/2022, 19:36:25][Debug] Loaded extended project 'undefined'
[29/08/2022, 19:36:41][Debug] Message from settings view: {"subject":"Workbench","index":0}
[29/08/2022, 19:36:41][Debug] Toolchain: selected 'Embedded Workbench 8.5 (MSP430)' (index 0)
[29/08/2022, 19:36:41][Debug] Generating intellisense config for 'MSP430F5257_clocktest':'Debug'...
[29/08/2022, 19:36:41][Debug] Loaded thrift workbench 'undefined'
[29/08/2022, 19:36:41][Debug] Not loading project 'MSP430F5257_clocktest' using thrift, no appropriate workbench selected...
[29/08/2022, 19:36:41][Debug] Loaded extended project 'undefined'
[29/08/2022, 19:36:41][Error] Didn't find a compiler for MSP430 in c:\Program Files (x86)\IAR Systems\Embedded Workbench 8.5.
[29/08/2022, 19:36:42][Debug] Intellisense config changed. Notifying cpptools.
[29/08/2022, 19:36:54][Debug] Providing intellisense configuration(s) for: e:\MSP430F5257_clocktest\main.cpp
[29/08/2022, 19:36:54][Debug] Generated intellisense configuration(s) in 5 ms.
[29/08/2022, 19:41:10][Debug] Message from settings view: {"subject":"AddWorkbench","index":0}
[29/08/2022, 19:41:24][Debug] Collected 0 toolchains
[29/08/2022, 19:41:38][Debug] Message from settings view: {"subject":"AddWorkbench","index":0}
```

error message is generated here:

https://github.com/IARSystems/iar-vsc-build/blob/8e665f11b6d94c146dcbe305f9b5725e0dfaf1c8/src/extension/cpptools/configurationprovider.ts#L272-L283

toolchainId is set here

https://github.com/IARSystems/iar-vsc-build/blob/01a04f763b0a3b9ace31e45eb7d64bdbf912b0b9/src/iar/project/parsing/xmlconfig.ts#L36-L41

toolchain name is parsed from project EWP file: 

https://github.com/IARSystems/iar-vsc-build/blob/01a04f763b0a3b9ace31e45eb7d64bdbf912b0b9/src/iar/project/parsing/ewpfile.ts#L68-L69

entry in EWP file

```
[...]
        <toolchain>
            <name>MSP430</name>
        </toolchain>
[...]
```

HampusAdolfsson, 2022-08-31:
Thanks for the detailed description!
You are correct, the code should not assume that the toolchainId is the same as the name of the toolchain folder.
I'll make sure we fix this for the next release.

HampusAdolfsson, 2022-10-07:
This is now fixed.

closed 2022-10-07 by HampusAdolfsson