iarsystems/iar-vsc-build #33: Duplicate Groups Created When Adding Files to Project

cookthebook, 2023-01-04:
I have been using the IAR VS Code extension since my project upgraded to IAR 9. I noticed that if I add a file to the project in VS Code, for any group at the same level as the new file, a new empty group with the same name also gets created.

[image]
[image]

This happens regardless of the alphabetical order of the new file:

[image]

I am using IAR 9.0 and v1.20.1 of the IAR Build extension on Windows 10.

cookthebook, 2023-01-04:
I was looking through the closed issues, and this might be a duplicate of https://github.com/IARSystems/iar-vsc-build/issues/12 ?

I can edit the project perfectly fine in the IAR Embedded Workbench, seems strange to me that there would be a disparity between the tools.

HampusAdolfsson, 2023-01-10:
I've added a workaround for the bug, which will be included in the next release. It should fix all cases of duplicated groups.

Note that the bug also causes you to be unable to remove any groups or files from a project in VS Code. There's nothing we can do about it, but there's now an error message to explain *why* groups/files aren't being removed.

cookthebook, 2023-01-10:
Thank you! I will watch for the next release.

commit 2b74d3ea (2023-01-10): VSC-122 Mitigate duplicated groups for EWARM 9.20

Closes #33
closed 2023-02-02 by IAR-build-bot