iarsystems/iar-vsc-build #35: Support workspace (.eww) files, batch builds, stronger Intellisense

gcampbell-msft, 2023-02-02:
Note: This may be a slight duplicate of #30, but I believe it's different enough for its own issue.

Support for workspace files would be make this extension much more like the IAR IDE, which would make it easier to use VSCode for the inner development loop. This would make it more obvious to use overall.

I believe that supporting workspace would make the following things available:
1. Easier viewing of a project from the IAR IDE, since .eww files are often (if not always) used in the IDE. 
2. Stronger Intellisense. Currently if a file isn't in the project the Intellisense could be much better. Expanding the Intellisense to files inside of the entire workspace, rather than just the active project, would be great. 
3. Batch builds. I think this would start down the path of supporting batch builds, which is something that is very helpful for users. 

Another note: It may make sense to split these into separate issues, but I created it as one for now.

schmitzel76, 2023-02-22:
Could definitely use this. We use batch build to make sure the projects are build in the correct order, but with vscode we now have to manually select and build each project.

leihen0525, 2023-04-07:
log:
`...
Updating build tree...

0  file(s) deleted.
Updating build tree...

 ERROR, Unknown argument variable: Failed to expand argument variable "$WS_DIR$"`

I have some public files that need workspaces as paths to boot, but VSC IAR doesn't support EWW, it would be nice if I could parse EWW

HampusAdolfsson, 2023-04-12:
@leihen0525 Workspace support is one of the features we're looking to implement next, but we can't give an exact date for it yet.

In the meantime, you can manually specify the value of `$WS_DIR$` by creating a file with the extension ‘.custom_argvars’ somewhere in your VS Code workspace with the following contents:

```xml
<?xml version="1.0" encoding="UTF-8"?>
<iarUserArgVars>
    <group name="MyGroup" active="true">
        <variable>
            <name>WS_DIR</name>
            <value>$PROJ_DIR$</value>
        </variable>
    </group>
</iarUserArgVars>
```

This file should be picked up and displayed in the extension configuration under “Custom Argument Variables File”.
This will have `$WS_DIR$` take the value of the project directory (the .ewp file directory), but you can change the value to whatever you need. It should solve any problems you have with building or generating intellisense configurations.

leihen0525, 2023-04-12:
@HampusAdolfsson This method works. Thank you very much for your help. Thank you!!!

HampusAdolfsson, 2023-06-26:
Workspace and batch build support was just released in 1.30.1! Please open new issues if you have any feedback or problems with it.

closed 2023-06-26 by HampusAdolfsson