iarsystems/iar-vsc-build #37: CMSIS Package Manager is opened when loading Git submodules

Spaark84, 2023-02-21:
I'm working on a project with several Git submodules, some of which contain an IAR project (.ewp file). When the submodule are loaded after opening VS Code, the IAR Build extension opens the CMSIS package manager for these projects and closes it again shortly after. This can happen multiple times in a row. It seems to happen on other occasions as well, but the reason is not clear to me. One can prevent this behavior by adding the .ewp files to the _files.exclude_ setting of VS code. I have this problem with IAR 9.32.1, IAR Build extension 1.20.3 and VS Code 1.75.1.

The IAR Build extensions should not open the CMSIS Package Manager automatically. It also would be nice if the IAR Build extensions would excluded the projects listed in .gitignore, so one can use patterns like the follwing, which currently are not possible in both the _files.exclude_ and the _iar-build.projectsToExclude_ settings

# Exclude all projects except the application
```
/**/*.ewp
!/**/application.ewp

```

jlonnberg, 2023-02-22:
The problem you're describing is a known problem when using CMSIS Enabled projects. The VS-Code plugins rely on our commanline utility iarbuild to generate the compile commands used for the project which unfortunatly triggers the Pack Manager to open for a short period of time. 

We'll investigate if we can suppress the behaviour.

jlonnberg, 2026-09-29:
The CMSIS Manager has been discontinued, so no more work will be done

closed 2026-09-29 by jlonnberg