iarsystems/iar-vsc-build #39: Command to get path to the built executable

axel-lebourhis, 2023-02-27:
Hi,

CMake Tools extension provide a command to retrieve the path to the executable: ${command:cmake.launchTargetPath}
It would be great to have a similar feature for this IAR extension, as it helps to create generic launch configurations, shared among teams.

Thanks a lot !

HampusAdolfsson, 2023-03-01:
Hi! What executable are you trying to find the path to? If it is an executable in the Embedded Workbench installation directory, we provide the `${command:iar-config.toolchain}` command which resolves to the selected Embedded Workbench or Build Tools installation.

For example:
`${command:iar-config.toolchain}/common/bin/iarbuild.exe` gives you the iarbuild executable, and `${command:iar-config.toolchain}/arm/bin/iccarm.exe` gives you the Arm compiler.

axel-lebourhis, 2023-03-01:
Hi ! No, I meant the output executable from the build.
The idea is to have a command that returns the absolute path to the .out of the currently loaded ewp, this way we can have generic launch configuration shared among teams.

HampusAdolfsson, 2023-03-01:
I see. For various reasons this is a bit complicated (but not impossible) to implement. If you haven't changed the output path from the default, you can get the path to the .out file like this:
`${command:iar-config.project-file}/../${command:iar-config.project-configuration}/Exe/${command:iar-config.project-name}.out`

Would that work for you?

axel-lebourhis, 2023-03-06:
I tried your suggestion, it does the job for us ! Thanks for the tip.

closed 2023-03-07 by HampusAdolfsson