iarsystems/iar-vsc-build #40: Include paths not set up to use clang-tidy

jwbmwv, 2023-03-01:
I like to use clang-tidy as my static analysis tool inside VS Code. This extension does not initialize the include paths that are needed for it, or any other extension needing paths to work properly since the proper header files annot be found.

jlonnberg, 2023-03-08:
@jwbmwv Hi, are you using a vscode-extension for Clang-Tidy and if so, which one? To me it seems like they typically work using a compilation database, much like the clangd extension. You can generate a compilation database using iarbuild and ninja but that will unfortunatly not be compatible with clang.

jwbmwv, 2023-03-10:
I'm using notskm.clang-tidy

I have no proof, but it appears that iar-build does not expose the include paths in a manor that this extension and/or clang-tidy itself can consume. The result is a list of include errors in the Problems tab for every file I open. This makes me ignore the problem window and miss useful messages. I have not had problems.

This appears to happen with any extension of this sort that does not generate a c_cpp_properties.json file automatically. i.e. Microsoft's Embedded Tools when importing an STM32Cube project.

jlonnberg, 2023-03-13:
Reading up on https://clang.llvm.org/extra/clang-tidy/ refers to the compilation database, from which the clang typically sources its information. I glanced through the source code for notskm.clangtidy and it support adding it supports adding options to the command line under clang-tidy.compilerArgs so you could possibly append the include paths there.

javari99, 2024-04-09:
Hello, I have been setting up clang-tidy using the c/c++ extension from vscode as @spoorcc suggested. The problem is that the extension is always trying to analyze the IAR compiler's standard lib source files and giving errors. How can I exclude such files?

Example message: `"call to undeclared function '__iar_something'; ISO C99 and later do not support implicit function declarations"`