iarsystems/iar-vsc-build #50: __FILE__ macro use different case of drive letter in iar-vsc-build and iar workbench

wsndmyr, 2023-06-12:
In vscode ```__FILE__``` use lower case of driver letter like "d:" while in iar workbench use upper case like "D:". 
This could cause a verification error when build again in vscode and then **debug without downloading** in iar workbench.

HampusAdolfsson, 2023-06-15:
Hi!

Could you clarify where in VS Code you see `__FILE__` with a lower case drive letter (e.g. in the intelliSense, or in the compiled program)? 

VS Code and Embedded Workbench build the program in the same way, so the value of `__FILE__` should always be the same in the compiled program.

wsndmyr, 2023-06-16:
In the output .bin file. 
I don't know much about vscode extension developing, I guess iar-vsc-build pass a lower case path to iarbuild.exe. Paths in .map file are different in driver letter when build with iar-vsc-build and iar workbench.

HampusAdolfsson, 2023-06-16:
This sounds like it might be an issue with iarbuild.exe, but I can't reproduce it. What version of Embedded Workbench are you using? Do you see this issue with a specific project, or any project you build with iar-vsc-build?

wsndmyr, 2023-06-19:
My iar version is 9.32.1.

I attach a simple project([Test.zip](https://github.com/IARSystems/iar-vsc-build/files/11783186/Test.zip)). It just print `__FILE__` and can be debugged with the simulator. 

When debug in iar workbench, the output is like:
[image]

When build with iar-vsc-build and then debug in vsc, the output is like:
[image]

And the terminal output:
[image]
It turns out iar-vsc-build do use a lower case driver letter.

HampusAdolfsson, 2023-06-19:
Thanks! This appears to be a bug in `iarbuild`, where the drive letter casing of the project path affects the paths used internally when compiling the project (if you run the `iarbuild` command yourself but change the path to `D:\Test\project.ewp`, it should give you an uppercase drive letter in `__FILE__`). 
The bug does not affect the latest EWARM 9.40.1.

We'll release a fix in the extension that ensures VS Code uses the same drive letter casing as Embedded Workbench.

HampusAdolfsson, 2023-06-27:
Sorry, there seems to have been a small mistake that caused the fix to not work as intended.
We'll push a bugfix release soon and make sure this works properly.

HampusAdolfsson, 2023-07-13:
We've just released another fix for this, please let us know whether it works for you and sorry again for the inconvenience.

wsndmyr, 2023-07-14:
Yes it works. Thanks a lot for your work!

closed 2023-07-14 by HampusAdolfsson