iarsystems/iar-vsc-build #53: AVR defines not recogognized by intellisense

brnhs, 2023-06-29:
I'm working with the VScode extension IAR Build V1.30.1 and can build all my projects.
But I have some problems with the c/c++ extension.

My configuration in c_cpp_properties.json looks like that.

```
        {
            "name": "AVR with IAR build",
            "configurationProvider": "iarsystems.iar-build"
        },
```

In my source file I have this line.

```
void main( void)
{
    __disable_interrupt();

}
```

The C/C++ estension reports a problem with the expression `__disable_interrupt();`

expected an expression
```
[{
	"resource": "main.c",
	"owner": "C/C++: IntelliSense",
	"code": "29",
	"severity": 8,
	"message": "expected an expression",
	"source": "C/C++",
	"startLineNumber": 956,
	"startColumn": 25,
	"endLineNumber": 956,
	"endColumn": 26
}]
```

Other example:

```
#pragma vector = TIMER0_COMPA_vect
__interrupt void TIMER0_COMPA_interrupt( void )
{
    TCNT0 = 0; 
    
}
```

identifier "TCNT0" is undefined
```
[{
	"resource": "/timer.c",
	"owner": "C/C++: IntelliSense",
	"code": "20",
	"severity": 8,
	"message": "identifier \"TCNT0\" is undefined",
	"source": "C/C++",
	"startLineNumber": 61,
	"startColumn": 5,
	"endLineNumber": 61,
	"endColumn": 10
}]
```

What is wrong?

Do you need more information?

HampusAdolfsson, 2023-06-29:
Hi!

Your first problem with `__disable_interrupt` is a bug. It is incorrectly declared in the extension and we will fix this in the upcoming bugfix release. As a workaround, you can:
* Go to the extension settings.
* Find the setting called `Iar-build: Defines`.
* Add the following line at the bottom: `__disable_interrupt=()`

Your second problem is caused by the device header files (e.g. `ioavr.h`). Some device headers use the IAR `@` operator to declare registers. Since `@` isn't standard C, VS Code's intelliSense doesn't understand the register declaration.

Unfortunately this isn't something we can fix until the VS Code C/C++ extension adds better support for IAR compilers. As a workaround, you can trick intelliSense to recognize the register by going to the `Iar-build: Defines` setting mentioned above and adding a line like this:
```
TCNT0=(*((char*)0))
```

brnhs, 2023-06-30:
Hi

Do get the workaround for `__disable_interrupt` working correctly, I needed to add to the defines this line `__disable_interrupt()=((void*)0)`.
Same for other IAR specific functions like `__enable_interrupt()` or `__no_operation()`.

I was able to define all IAR registers as described.

Thanks

commit 5d3f7c8f (2023-06-30): VSC-413 Update intrinsic function blacklist

Closes #53
closed 2023-07-13 by IAR-build-bot