iarsystems/iar-vsc-build #54: 1.30.1 fails to load projects

davidanderle, 2023-07-03:
Since the rollout of the latest 1.30.1 version, my projects fail to load in the "IAR Build" tab with the following error:
[image]
[image]
[image]
Rolling back to 1.20.3 resolves it. Have you got any ideas how I could trace down the problem? For now, i'll just stick to the previous version.

HampusAdolfsson, 2023-07-04:
Hi!
Could you please take a look at the extension logs? You can find them in the "Output" panel next to the Debug Console. Select "Iar Build" in the dropdown on the right side.
When everything goes correctly, there should be entries that look like this (but with different names):
```
[7/4/2023, 10:56:46 AM][Debug] Loaded thrift workbench '9.30.1 (Arm)'
[7/4/2023, 10:56:46 AM][Debug] Loading thrift workspace: blink_led
[7/4/2023, 10:56:48 AM][Debug] Successfully loaded 1 projects in workspace
[7/4/2023, 10:56:48 AM][Debug] Loading blink_led
[7/4/2023, 10:56:48 AM][Debug] Loaded extended project 'blink_led'
```
Which of these can you see? The project manager is most likely crashing somewhere in this process, and it would be helpful to know when.

davidanderle, 2023-07-04:
Hello,
Thanks for looking into it. Sorry, based on the contributing guidelines it was not clear where to find this output (i was missing the fact that there was a dropdown!)
This is the IAR Build:
[image]

And this is the IARServiceManager:
[image]

dnjd4219, 2023-07-04:
Same problem.
I don't know if this will help, but here are the extended logs.
[image]
[image]

HampusAdolfsson, 2023-07-05:
Thanks for the logs!

This is a bug in EWARM 9.30 that causes the project manager used by VS Code to crash whenever there is a cycle in the project tree, for example like this:
[image]

It was fixed in EWARM 9.40. 

To fix this, you can remove the `.dep` file with the same name as your project (`.ewp`) file and reload VS Code.

The `.dep` file is recreated when you build the project in Embedded Workbench and then close it, but in my case the new `.dep` file doesn't cause any problems. Let me know if the crash comes back for you after doing this. In that case we can work on a better fix in the extension.

davidanderle, 2023-07-05:
Hi @HampusAdolfsson, 

Thanks, I can confirm that removing the .dep file works which is an OK workaround for me. Unfortunately right now we cannot justify spending time assessing the update to 9.40 so I'm stuck with this for now. 

Could you please highlight how and why cycles are created in the project?

Lastly, can you please point me to a comprehensive release log for EWARM 9.40 and for IAR Build 1.30.1 version as I did not see any "VS Code integration improvement".

HampusAdolfsson, 2023-07-05:
@davidanderle Great!

Cycles do not usually occur. They may appear when upgrading a project from an older version, but I haven't found any other case where they appear. Removing the `.dep` file or performing a rebuild should get rid of them.

We release a few bugfixes and improvements to the VS Code integration with each release of Embedded Workbench, but we typically do not mention them in the Embedded Workbench release notes. Instead, we list significant bugs/limitations that affect specific EW versions in [the extension README](https://github.com/IARSystems/iar-vsc-build#compatibility).

For this bug specifically, it originally did not affect any users (hence it is not mentioned anywhere). It was only with the new workspace support in IAR Build 1.30.1 that this bug started triggering.

You can find the release notes for the IDE component that the VS Code integration uses here:
* [https://netstorage.iar.com/FileStore/STANDARD/001/002/595/arm/doc/infocenter/ewarm.ENU.html](https://netstorage.iar.com/FileStore/STANDARD/001/002/595/arm/doc/infocenter/ewarm.ENU.html)

And for the extension here:
* [https://github.com/IARSystems/iar-vsc-build/releases](https://github.com/IARSystems/iar-vsc-build/releases)

davidanderle, 2023-07-05:
@HampusAdolfsson Thanks for the links and for the clarification!

dnjd4219, 2023-07-06:
@HampusAdolfsson 
As you said, if I remove the .dep file and run vscode, it's fine.
However, if I build the project in vscode and then run vscode again, I get the same error. 

Should I remove the .dep file and run vscode every time?

HampusAdolfsson, 2023-07-06:
@dnjd4219 That sounds like it would be annoying to do every time.

If you open the project in Embedded Workbench, can you find any cycles or loops in the project tree, like the one in the image I sent above? I would like to know what is causing it to appear every time you build.

In any case, you can try a more aggressive workaround. In your workspace folder, create the file `.vscode/settings.json` if it does not exist, and add the following content:
```json
{
    "files.exclude": {
        "**/*.eww": true
    }
}
```
This will prevent the extension from loading workspace files, and should fix the crash. It also means that you won't be able to load/save batch builds from the workspace file, so if you use batch builds this might not be the best solution.

dnjd4219, 2023-07-07:
@HampusAdolfsson 
Yes.
I found the loop in my project.
[image]

I added that exclude setting to my settings.json and it works fine.
But I don't think this is the best solution.

closed 2024-01-25 by IAR-build-bot

HampusAdolfsson, 2024-01-25:
The newest version (1.30.5) includes a workaround for this. Before loading a project with EWARM 9.30 (and other affected versions), the extension removes the .dep file automatically.

Let us know if you experience any more issues with this!