iarsystems/iar-vsc-build #55: Error initiating IAR toolchain backend. Some functionality may be unavailable (Error: Service registry exited prematurely, code: 1).

fubinfeng2333, 2023-07-25:
I used the iar8.10.1 version to create the project before, and then I installed the iar9.40.1 version today to open the previous project, make it into the 9.40 version, and then use the iar build plug-in. At the beginning, it can still display the file structure, but then I don’t know why it doesn’t show it. Please take a look for me, thanks.
[image]
[image]
[image]

jlonnberg, 2023-08-07:
Hi!
Can you try opening the project in EWARM once and save the project to ensure that it's been properly upgraded before trying it out in VSCode again?

BR
Joakim

fubinfeng2333, 2023-08-09:
I made sure that I have converted the iar8.10 project to 9.40, please see the picture below, when I open the old project with the 9.40 version, it pops up these windows, and I have selected yes, even I have compiled the project.
[image]
[image]

jlonnberg, 2023-08-09:
Can you include the log from IAR Build for me when the problem occurs? It can be obtained from the output tab and select "IAR Build" in the dropdown menu.

Moreover, are you able to build the project properly usign the iarbuild build task? If so, can you please include the build output as well.

fubinfeng2333, 2023-08-15:
The following screenshot is the log of IAR build:
[image]
[image]

jlonnberg, 2023-08-17:
Does the path to the project contain any non-ascii characters? Moreover, is it possbile for us to get the project to test internally?

jlonnberg, 2023-08-22:
Please also consider if the suggested fix for #58 might fix this problem as well

fubinfeng2333, 2023-09-01:
I updated iar build to v1.30.3, but found the same problem.
[image]

fubinfeng2333, 2023-09-01:
It seems that this problem will occur whenever the iar project is converted from version 8.0 to version 9.0 or above. I tried it with the project (EWARM V8) generated by stm32cube, and the result was the same.
[image]

jlonnberg, 2023-09-06:
I've just tried upgrading a project from 8.10 to 9.40 and was able to show the view. To be able to identify the problem, I could really use the project that fails. Can you share it with me?

fubinfeng2333, 2023-09-07:
[cube_test_tttt_0901_2.zip](https://github.com/IARSystems/iar-vsc-build/files/12545307/cube_test_tttt_0901_2.zip)
This is the project I generated with the stm32cube tool.

jlonnberg, 2023-09-07:
The supplied project load perfectly for me and all the files are visible. Looking at your logs shows that your installation seems to have multiple compilers in the arm\bin directory. Is that the case? Does your installation for some reason contain duplicates of any other binaries?

fubinfeng2333, 2023-09-07:
How should i check？

fubinfeng2333, 2023-09-07:
Most of the files are different, only a few files are the same.
[image]
[image]

jlonnberg, 2023-09-07:
So, can you supply a list of all files matching "icc*.exe" from D:\Program Files\IAR Systems\Embedded Workbench 9.2\arm\bin

fubinfeng2333, 2023-09-07:
[image]

jlonnberg, 2023-09-07:
It seems like your installation is corrupt, it lists multiple backup files (.bak) which should not be present. I suggest that you remove the installation and start with a fresh installation. That should fix the problem

closed 2023-09-08 by jlonnberg