iarsystems/iar-vsc-build #56: [IAR 9.20]Could not parse workspace file...

Flymachine, 2023-07-25:
I got a error from IAR Build:
```
[7/25/2023, 3:49:15 PM][Error] Could not parse workspace file 'd:\demo\demo.eww': Error: Expected root tag to be 'workspace', not 'undefined'
```
The project works well with IAR 9.20.

It's the content of .eww file:
```
<?xml version="1.0" encoding="iso-8859-1"?>
<!-- This file was generated by Simplicity Studio -->
<workspace>
  <project>
    <path>$WS_DIR$\demo.ewp</path>
  </project>
  <batchBuild/>
</workspace>
```

I found that if I delete the comment in .eww file as this:
```
<?xml version="1.0" encoding="iso-8859-1"?>
<workspace>
  <project>
    <path>$WS_DIR$\demo.ewp</path>
  </project>
  <batchBuild/>
</workspace>
```
No error reported. 

That means your xml parse module has a bug.

Flymachine, 2023-07-25:
Another question: 
I tried to set the config to exclude sdk projects:
```
"iar-build.projectsToExclude": "sdk"
```
However, I still could see the related projects in IAR Build toolbar.
For example: `d:\demo\gecko_sdk\platform\CMSIS\RTX\RTX_CM.eww`. 
They aren't ignored.

I tried to set it as:
```
"iar-build.projectsToExclude": ".*sdk.*"
```
It still didn't work.

How to set this config to ignore those projects?

jlonnberg, 2023-08-07:
Hi,

A fix for the comment problem will be released shortly.

jlonnberg, 2023-08-07:
Regarding the sdk projects; are you using working with an active workspace file or in folder without any .eww file?

Flymachine, 2023-08-07:
Our project has a .eww file at the same folder as .ewp file.  However, the SDK library we use also has some eww and ewp files. I want to find a way to ignore the root path of the SDK

jlonnberg, 2023-08-09:
Regarding the filtering problem; The filter only acts on projects and not on workspaces. Moreover, the filter function is only active when working in a folder without an eww file. If an eww-file is selected, no filtering is applied and all the projects listed in the workspace will be shown in the dropdown menu for available projects.

Flymachine, 2023-08-09:
I see.  I misunderstood. Thanks. This issue could be closed now.

closed 2023-08-24 by jlonnberg