iarsystems/iar-vsc-build #57: Increasing the (re/)build task verbosity level

felipe-iar, 2023-08-07:
The `build` and `rebuild` tasks comes with ` '-log' 'info'` hardcoded in the command line:
```
'C:\\...\\9.40.1.63915/common/bin/iarbuild.exe' 'D:\prj\prj.ewp' '-build' 'Debug' '-log' 'info' '-varfile' ...
```
When trying to raise verbosity to `-log all` in the rebuild task:
```json
{
	"version": "2.0.0",
	"tasks": [
		{
			"type": "iar",
			"command": "rebuild",
			"project": "${command:iar-config.project-file}",
			"config": "${command:iar-config.project-configuration}",
			"builder": "${command:iar-config.toolchain}/common/bin/iarbuild.exe",
			"label": "iar: Rebuild Project",
			"argumentVariablesFile": "${command:iar-config.workspace-file}",
			"extraBuildArguments": ["-log", "all"],
			"problemMatcher": [
				"$iar-cc",
				"$iar-linker"
			]
		}
	],
}
```
There is a clash:
```
'C:\\...\\9.40.1.63915/common/bin/iarbuild.exe' 'D:\prj\prj.ewp' '-build' 'Debug' '-log' 'info' '-log' 'all' '-varfile' ...
```
... and `-log info` wins.

Is there any way to override the default behavior to maximize verbosity (`-log all`)?

jlonnberg, 2023-08-09:
Hi

You can change the verbosity in the settings for the iarbuild extension settings; search for iar-build.buildOutputLogLevel and you can alter it to the level you want.

closed 2023-08-09 by jlonnberg

felipe-iar, 2023-08-09:
Found it. Thanks @jlonnberg. 

[image]