iarsystems/iar-vsc-build #61: Can't load project with multiple prebuild directives

mfruendtxylem, 2023-10-04:
**Describe the bug:**
When I load a project with multiple prebuild directives like

<prebuild>python "$PROJ_DIR$\CommonCode\version_gen.py" chdr -o "$PROJ_DIR$\CommonCode\version.h" --dirty && $PROJ_DIR$\Tools\AStyle\astyle.bat</prebuild>

I get the following error

Could not parse project file 'c:\Users\mfruend\Documents\Projekte\EcoTouch_UV\EcoTouch_UV.ewp': Error: Invalid character in entity name
Line: 6577
Column: 124
Char: &

Loading the project in IAR works fine.

**To Reproduce:**

Have multiple prebuild directives, connected by && in the project.

**Expected behavior:**

The project should load, just like in IAR

**Actual behavior:**

Loading is aborted

**Environment:**

- OS: Windows
- Embedded Workbench: Arm
- Embedded Workbench version: 9.32
- VSC Extension version: I Tested 1.30.3 and 1.30.2

**Additional context:**
Add any other context about the problem here, e.g., log, projects, files etc. Logs can be obtained from the output view under the "IarServiceManager", "IAR Build", "IAR Config Generator" and "C/C++" sections.

mfruendtxylem, 2023-10-04:
After finally building with IAR I noticed IAR automatically replaced `&&` with `&amp;&amp;` in the ewp file and after that it works again. So not sure if this is something that can be fixed on your side. Possibly the error could tell the user to replace `&&` with `&amp;&amp;` or the auto conversion is also applied.

jlonnberg, 2023-11-28:
Hi @mfruendtxylem,
Did you manually add the pre-build to the ewp-file or did you add it using the Embedded Workbench. We typically discourage users from manually editing the ewp-file as we cannot guarantee that the file will be readable by the IDE and/or VsCode.

mfruendtxylem, 2023-11-28:
I'm mostly sure it was through EW, but an older version at the time, though we also used https://github.com/IARSystems/project-migration-tools at the time which messed with the `&&` too (I was told it's fixed now). So it mostly confused me that the IAR compiler did automatic correction while the VSC extension did not.

jlonnberg, 2023-11-29:
So, we do some minor parsing that differs between the Embedded Workbench and VsCode which will be removed in the next release. So most likely, this will be fixed in the next release!

closed 2024-11-26 by HampusAdolfsson