iarsystems/iar-vsc-build #64: Cannot build single file at a time

fenwics, 2023-11-08:
When I have a project loaded and I right click on the tab for the active file and select "Compile Active File", I get the error message "Unable to find compilation information for this file". See below for screenshots. I am using IAR EWARM v9.0_3 with IAR Build Extension 1.30.4 and VS Code 1.83.1.

[image]

[image]

HampusAdolfsson, 2023-11-14:
Hi!

The "Compile Active File" option is provided by the [CMake Tools extension](https://marketplace.visualstudio.com/items?itemName=ms-vscode.cmake-tools), and as you might expect it does not work with Embedded Workbench projects.

The `IAR Build` extension doesn't support building single files at the moment. If you'd like, we can consider this a feature request.

jaegerbeat, 2025-09-09:
[image]
Something like this would be incredibly useful