iarsystems/iar-vsc-build #66: Build fails when using the $WS_DIR$ macro in the prebuild command

integerfn, 2023-12-29:
**Describe the bug:**
Build fails with error: `ERROR, Unknown argument variable: Failed to expand argument variable "$WS_DIR$"` when building from Visual Studio Code, but works on IAR. It happens when the $WS_DIR$ macro is used in the "Pre-build command" field, within "Build Actions".

I want to run a command as a prebuild action and, due to the directory structure, it's better to use the WS_DIR macro rather than $PROJ_DIR$. 

**To Reproduce:**
Build the project in Visual Studio Code

**Expected behavior:**
Successful build

**Actual behavior:**
Build fails.

**Environment:**

- OS: Windows 10
- Embedded Workbench: Arm 
- Embedded Workbench version: 9.30.1.50054
- VSC Extension version: v1.30.4

HampusAdolfsson, 2024-01-03:
Hi,

I can't reproduce this on my end (I tested with the prebuild command `echo '$WS_DIR$`). I have two ideas that could help narrow down the issue:

* Does VS Code detect the workspace file? Under "Extension Configuration" in the "IAR Build" panel, does the workspace dropdown contain your workspace (`.eww` file)?
* Is there a `.custom_argvars` file passed to the `iarbuild` process? When you build the project in VS Code, the command line is printed at the start of the build log. It should contain something like this:
```'-varfile' 'C:\Users\user\AppData\Local\Temp\iar-build\08f0bc4a-6306-4282-b766-a07b0aa83feb.custom_argvars'```
This file is used to pass the value of `$WS_DIR$` to the builder.

integerfn, 2024-01-03:
Hi, 

- VS Code does detect the workspace file. I have 2 projects within the same `.eww` with two configurations (Debug and Release) each. I can see them in the dropdown in the IAR Build panel. However, if I attempt to switch project, VS Code crashes, but that seems like a different issue.

- There is a `.custom_argvars` file that I'm using to interface with a CI/CD pipeline, so it just has the serial number for the device and a few more compile constants.  The command line printed at the beginning of the console in VS Code is something like this:

`'C:\Program Files\IAR Systems\Embedded Workbench 9.1 for ARM v9.30/common/bin/iarbuild.exe' 'c:\Users\.......\myproject.ewp' '-make' 'Debug' '-log' 'info' '-parallel' '4' '-varfile' 'c:\Users\.......\myfile.custom_argvars'`

What you explained seemed like the VS Code IAR Build extension is creating a temporary `.custom_argvars` file to pass the `$WS_DIR$`, but if I'm supplying my on file, would that prevent that mechanism from working?

HampusAdolfsson, 2024-01-04:
The extension creates a temporary `.custom_argvars` file that is a copy of your `.custom_argvars` file, but which also defines the `$WS_DIR$` variable. This is so that you can use `$WS_DIR$` in your projects, even though IAR Build does not yet support workspaces. Your own argument variables should still work.

Are there any unusual paths involved (e.g. symlinks or network drives)? These can cause issues if they are handled differently by VS Code and the IDE.

In any case, you can try overriding the `.custom_argvars` file the extension uses:
1. Create a copy of your `.custom_argvars` file that you can use for experimenting.
1. In VS Code, go to `Terminal->Configure Tasks...` and select `iar: Build Project`. 
1. In the tasks.json file, replace the value for `argumentVariablesFile` with the path to your new `.custom_argvars` file. You might also want to change the `label` to something else.

If you run this task (`Terminal->Run Task...`), it will use the `.custom_argvars` file you provided instead of creating a temporary one. You can experiment with setting your own value for `$WS_DIR$` like this (in the `.custom_argvars` file):
```xml
<?xml version="1.0" encoding="UTF-8"?>
<iarUserArgVars>
    <!-- other groups... -->
    <group name="MyGroup" active="true">
        <variable>
            <name>WS_DIR</name>
            <value>PATH_TO_WSDIR</value>
        </variable>
    </group>
</iarUserArgVars>
```

If you can find out what values work or don't work, I'd be very interested to know.

integerfn, 2024-01-05:
If I add the `WS_DIR` variable to the `.custom_argvars` file VS Code build works correctly. It looks like the temporary copy is not being done. The command line in VS Code shows the path to my file, not a file under `temp` like you said. 

There are no funny things with the paths, like symlinks or network drives. The only thing slightly out of the ordinary is that the path has spaces.

What could cause the VSCode extension not to create the temporary copy?

integerfn, 2024-01-08:
Thank you! That was the issue. For some reason the tasks.json had this `"argumentVariablesFile": "${command:iar-config.argument-variables-file}`. I changed it to `${command:iar-config.workspace-file}` and now it works. I don't recall setting that by hand, perhaps it's the default value from an older version?

Thank you!

HampusAdolfsson, 2024-01-09:
Great, happy to help! You're right, `${command:iar-config.argument-variables-file}` used to be the default before the extension had support for `.eww` files.

closed 2024-01-09 by HampusAdolfsson