iarsystems/iar-vsc-build #68: Intellisense fails for NAN and INFINITY from iar math.h for RX

larlin, 2024-02-12:
Intellisense reports error for NAN and INFINITY macro defined in iar math.h for RX.

**To Reproduce:**
`#include <math.h>

float test = NAN;
`

[image]

**Expected behavior:**
A clear and concise description of what you expected to happen.

**Actual behavior:**
What actually happens.

**Environment:**

- OS: [Windows, linux]
- Embedded Workbench: [Arm, RX, RL78, AVR, RISC-V, ...]
- Embedded Workbench version:
- VSC Extension version:

**Additional context:**
Add any other context about the problem here, e.g., log, projects, files etc. Logs can be obtained from the output view under the "IarServiceManager", "IAR Build", "IAR Config Generator" and "C/C++" sections.

HampusAdolfsson, 2024-02-12:
Hi,

The `0.NaN` literal used by the `math.h` header is an IAR C extension, and isn't understood by the C/C++ parser used in VS Code. However, you can try enabling more portable definitions with the `__RXABI_PORTABILITY_LEVEL` preprocessor define. In the extension settings, go to "Defines" and add this entry:
```
__RXABI_PORTABILITY_LEVEL=1
````
This will define it for VS Code's parser only, so it does not affect your build in any way.

larlin, 2024-02-13:
Great that worked thanks.

closed 2024-02-13 by larlin