iarsystems/iar-vsc-build #70: No command available to build current project

tvandijk-alflex, 2024-02-21:
There currently is no VS Code command available to build the current project.
Due to this it is not possible to add a build key binding in VS Code, instead requiring several steps in the GUI.
Is it possible for such a command to be added? It would result in a large UX improvement.

micaelbo, 2024-02-21:
Did you try setting the default Build Task via Ctrl+Shift+B and select "iar: build project" under "Configure Default Build Task" from the global Terminal menu?

tvandijk-alflex, 2024-02-21:
I had not tried that one yet, but that does seem to work.
Interestingly the "iar: build project" option does not show as an option up by default when trying to run the build task without one configured, but is available with the "Configure Default Build Task" command.