iarsystems/iar-vsc-build #72: Incorrect intellisense for currently selected project

mloccisa, 2024-03-07:
**Describe the bug:**
The preprocessor defines which are active in VS code intellisense do not match those of the currently selected project in the extension. Instead they match a different project in the same workspace.

**To Reproduce:**
1. In a new IAR workspace (EWW) file, create two empty projects (EWP files).
2. In the project options for the **first** project, C/C++ Compiler->Preprocessor->Defined symbols add `MY_DEFINE_1`.
3. In the project options for the **second** project, C/C++ Compiler->Preprocessor->Defined symbols add `MY_DEFINE_2`.
4. Create a new C file. In the file, add two conditional blocks, one for each of the above defines.
   ``` C
    #if defined(MY_DEFINE_1)
   // do something
   #endif
   #if defined(MY_DEFINE_2)
   // do something
   #endif
   ```
5. Add the C file just created to both of the projects and save all files. This won't yet compile, but that's not necessary for this reproduction.
7. Open the project in VS Code with the IAR extension installed.
8. Configure the IAR extension to the correct toolchain and select the project with only `MY_DEFINE_2` defined.
9. Open the C file you created in VS Code.
10. Observe that the intellisense is not correct for the given project. The block under `MY_DEFINE_1` is shown as active while the block under `MY_DEFINE_2` is shaded indicating it is inactive.

**Expected behavior:**
The active preprocessor defines in intellisense should match the active project selected in the extension.

**Actual behavior:**
The active preprocessor defines in intellisense match another project in the same workspace.

**Environment:**
[IAR Build extension bug.zip](https://github.com/IARSystems/iar-vsc-build/files/14530147/IAR.Build.extension.bug.zip)

- OS: Windows
- Embedded Workbench: Arm
- Embedded Workbench version: 9.10.2
- VSC Extension version: 1.30.6

**Additional context:**
A more complicated example with 3 projects is attached here:
[IAR Build extension bug.zip](https://github.com/IARSystems/iar-vsc-build/files/14530147/IAR.Build.extension.bug.zip)
From this we can see, files which are part of two projects can get intellisense from the wrong one, and the intellisense is not applied consistently across files - another file not part of either of those first two still gets intellisense applied from another, un-selected project. So we really don't know which project the intellisense information is coming from on any file we're viewing in VS Code.

HampusAdolfsson, 2024-03-08:
You are correct, the extension should prioritize picking the intellisense configuration from the currently selected project, if there are several projects the file belongs to. It currently picks a project arbitrarily. We'll fix that.

> another file not part of either of those first two still gets intellisense applied from another, un-selected project

This is the intended behaviour. It helps when editing files from several different projects simultaneously, since you do not have to change the selected project every time you switch to a file that belongs to a different project.

mloccisa, 2024-03-11:
In that case of a file that's not part of the selected project, what happens if it's part of two or more other projects? Will it just be picked arbitrarily from those other projects?

I agree it can be useful to show some intellisense for those files from another project. It would be good however to have some indication (I'm not sure how) of which project the intellisense is coming from without reading through the logs. Or possibly just indication that it's been chosen arbitrarily. That way if it was an issue for some user, at least they're aware the intellisense might not match what they expect and they can change their selected project to get accurate information for the file they're reading.

HampusAdolfsson, 2024-03-11:
Yes, it will be picked from one of the projects arbitrarily. A possible improvement would be to prioritize projects in order of most to least recently selected/active. At the moment that would be quite a bit of work for a relatively small gain, but we can keep it in mind for the future.

In my opinion, the logs are a fitting place to present the information about what project the intellisense information is generated from, with the caveat that it's difficult to find the logs (or to know to look for them).
We could add a notification (the kind that appears in the bottom right corner) whenever a conflict of this kind appears, to point them towards the logs, but that might be a bit obtrusive.

ciscoski, 2024-03-12:
+1 on this.
Thanks

HampusAdolfsson, 2024-11-26:
This has been addressed in version `1.30.7` by always using the IntelliSense configuration from the active project if possible. If a file belongs to multiple projects but none of them are the active one, we pick a project arbitrarily like before. Any feedback on how this works for you is welcome!

closed 2024-11-26 by HampusAdolfsson