iarsystems/iar-vsc-build #75: Sync launch.json with project files

davidanderle, 2024-04-05:
Is there a way to automatically sync the launch.json file with the IAR project files? At the moment if the user changes a debug setting in IAR when the launch.json file already exists, the change does not sync to the VSCode environment. I could create tools to parse the relevant files and update the launch.json on a trigger (eg before a Launch task), but it seems like i'd be reinventing the wheel... 

Since Run > Add configuration: IAR C-SPY Debug command can automatically create a basic template file, I was wondering if I could somehow access this "parsing tool". I would also like to add the extra "preLaunchTask": "iar: Build Project" key to the template file by default.

HampusAdolfsson, 2024-04-09:
> I would also like to add the extra "preLaunchTask": "iar: Build Project" key to the template file by default.

Adding an extension option called "Build before debug" (or something to that effect) definitely sounds like something we want to do. This would cause the `preLaunchTask` to be added by default, and would also take effect if debugging without a launch.json file.

For your main question, the *intended* solution is to not use a launch.json file. In the debug configuration dropdown, you can press "IAR C-SPY Debug..." to use the automatic debug configurations without creating a launch.json file:

[image]
When you go into the menu again (or restart VS Code), the configurations are regenerated from the project files.
I realize this has its issues and pain points, and doesn't work well if you need to customize the launch.json file in some way (for example to add a `preLaunchTask`). Would this work well enough for you?

Otherwise, we could look at allowing you to omit certain fields from a launch.json configuration, such as the `program` or `driverOptions`, and having the extension fill in missing fields automatically from the project files.

HampusAdolfsson, 2024-04-09:
Unfortunately, there is no command line interface to the VS Code debugger or our launch.json generator. I can't think of any other quick solutions to make this easier, either.

To clarify, with my suggestion above, you would be able to define a minimal launch.json configuration like this:
```json
{
    "type": "cspy",
    "request": "launch",
    "name": "Debug with C-SPY",
    "preLaunchTask": "iar: Build Project",
    "workbenchPath": "${command:iar-config.toolchain}",
    "projectPath": "${command:iar-config.project-file}",
    "projectConfiguration": "${command:iar-config.project-configuration}",
},
```
This just says to debug the project and configuration currently selected in the "IAR Build" extension GUI, but not how to debug. All the details, like the program path and driver, would be filled in from the project files every time you launch it.

davidanderle, 2024-04-09:
This would work for us I think, but I'm having this problem and therefore the debugger is not starting:
[image]

davidanderle, 2024-04-11:
Ah, I see! Thanks, we will wait for the new release

HampusAdolfsson, 2024-11-26:
This is now released in the latest extensions (`1.30.7` for IAR Build and `1.40.1` for IAR C-SPY Debug). Note that you need to update *both* extensions. The smallest possible launch.json configuration looks like this:
```json
{
    "type": "cspy",
    "request": "launch",
    "name": "Debug the active IAR Project with C-SPY",
    "workbenchPath": "${command:iar-config.toolchain}",
    "projectPath": "${command:iar-config.project-file}",
    "projectConfiguration": "${command:iar-config.project-configuration}"
}
```

closed 2024-11-26 by HampusAdolfsson

d-anderle, 2024-11-26:
Thank you @HampusAdolfsson! I am in progress of the testing the new features, and it appears to me that with the minimal configuration I still cannot upload the correct binary on the target for debugging. In my custom_argvars I have a revision definition. This controls the name of the output file, such as my_software_rev_$REVISION$.out. When I change this define, EWARM starts using the latest revision output file, but VSCode would still use the old one. Is there a way I could change this behaviour?

d-anderle, 2024-11-26:
@HampusAdolfsson thank you, the 2nd solution works well relatively fast, but the first one fails as the IAR project file for some reason is no longer recognised. Manually adding a new line works, so I think it could be because I'm on PS environment. In any case, I am happy to use this workaround until the window is automatically reloaded on changing the .custom_argvars.