iarsystems/iar-vsc-build #79: Intellisense problem with $CONFIG_NAME$ preprocessor

xNiux, 2024-05-07:
**Describe the bug:**

I add this option in each IAR configuration preprocessor : 

[image]

When you change the active configuration, the active/inactive code are not updated with intellisense

**To Reproduce:**
Update your IAR configurations (More than one) with $CONFIG_NAME$ in preprocessor defined symbols.
Open your project with VS Code, open C file with precompilation code.
Change configuration

**Expected behavior:**
Inactive code of new configuration become dark, active code of new configuration become light

**Actual behavior:**
No change on C files : 

[image]

[image]

**Environment:**

- OS: Widnows 10
- Embedded Workbench: ARM
- Embedded Workbench version: 9.20
- VSC Extension version: 1.30.6

**Additional context:**
Add any other context about the problem here, e.g., log, projects, files etc. Logs can be obtained from the output view under the "IarServiceManager", "IAR Build", "IAR Config Generator" and "C/C++" sections.

HampusAdolfsson, 2024-05-10:
I haven't been able to reproduce this. Does this happen for all preprocessor defines or only `$CONFIG_NAME$`?

Note that there was a bug that caused intellisense info to not be updated when changing configuration (#63). This was fixed in 1.30.6 on March 6. Could this be the bug you're experiencing, or have you seen this more recently too?

xNiux, 2024-05-13:
Problem seems to be on all pre-defined symbols in IAR C/C++ Compiler options : 
[image]

But directly in code it is ok : 
[image]

Problem is quietly recent. maybe 3 or 4 months

xNiux, 2024-05-14:
Ok, IAR Config Generator is OK. But for IAR Build, it missing the 2 last lines : 

At project open : 

[image]

At config change : 

[image]

HampusAdolfsson, 2024-05-17:
The "cpptools" mentioned in the logs is Microsoft's "C/C++" extension. It seems we successfully notify it about the intellisense change. It is then supposed to request updated configurations for all open files, but it doesn't.

Could you look in the *C/C++* log channel to see if there's anything interesting logged when you change configuration? This is what it looks like for me:
```
LSP: (received) cpptools/clearCustomConfigurations
LSP: (invoked) cpptools/clearCustomConfigurations
Intellisense update pending for: file:///d%3A/repos/VSCode/iar-vsc-debug/samples/GettingStarted/Fibonacci.c
Shutting down IntelliSense server: D:\repos\VSCode\iar-vsc-debug\samples\GettingStarted\Fibonacci.c
LSP: (received) cpptools/resumeParsing
LSP: (received) cpptools/queryTranslationUnitSource: file:///d%3A/repos/VSCode/iar-vsc-debug/samples/GettingStarted/Fibonacci.c (id: 39)
LSP: (invoked) cpptools/queryTranslationUnitSource: file:///d%3A/repos/VSCode/iar-vsc-debug/samples/GettingStarted/Fibonacci.c (id: 39)
Populating file name cache...
LSP: (invoked) cpptools/resumeParsing
Done populating filename cache. Elapsed time: 17 ms
LSP: Sending response (id: 39)
Custom configurations received:
```

You may need to first set the "Logging level" to Debug in the settings for the "C/C++" extension (you will get *lots* of logs with this setting):
[image]

You could also try running the command "C/C++: Log Diagnostics" from the F1 command palette. The output at the top should look something like this:
```json
Version: 1.20.5
Current Configuration:
{
    "name": "Win32",
    "includePath": [
        "d:/repos/VSCode/iar-vsc-debug/samples/GettingStarted/**"
    ],
    "defines": [
        "_DEBUG",
        "UNICODE",
        "_UNICODE"
    ],
    "windowsSdkVersion": "10.0.22621.0",
    "compilerPath": "cl.exe",
    "cStandard": "c17",
    "cppStandard": "c++17",
    "intelliSenseMode": "windows-msvc-x64",
    "compilerPathInCppPropertiesJson": "cl.exe",
    "intelliSenseModeIsExplicit": false,
    "cStandardIsExplicit": false,
    "cppStandardIsExplicit": false,
    "mergeConfigurations": false,
    "compilerPathIsExplicit": false,
    "configurationProvider": "iarsystems.iar-build",
    "browse": {
        "path": [
            "d:/repos/VSCode/iar-vsc-debug/samples/GettingStarted/**",
            "${workspaceFolder}"
        ],
        "limitSymbolsToIncludedHeaders": true
    }
}
```
The most important line here is `"configurationProvider": "iarsystems.iar-build",`, which means that this extension is actually used to provide intellisense configurations.

xNiux, 2024-05-17:
Hoho, I think problem is here : 

[image]

xNiux, 2024-05-17:
You're right ! Thanx a lot !

closed 2024-05-17 by xNiux