iarsystems/iar-vsc-build #81: Intellisense not working properly showing unrelated variables/select intelliSense config not working

justric123, 2024-07-26:
**Describe the bug:**
My simple class without inheritance shows a large list of unrelated variables that are not member variables. This problem occurs only in VSC but not in IAR IDE. Here only the variables in scope are available. 
All unrelated variables show the message if you click on details:
"Note: IntelliSense is not fully configured. Use the 'Select IntelliSense Configuration...' command to finish configuration.". But even if selecting a config it won't disapear. So I guess the selection step doesn't work properly. Maybe I can add the specific config directly in settings.json or iar-vsc.json?

**To Reproduce:**
any class I add has this behavioral. In any method/constructor/destructor.

**Expected behavior:**
Intellisense only shows member variables and methods of the specific class.

**Actual behavior:**
unrelated variables make it impossible to find members/methods.

**Environment:**

- OS: Windows 10 Pro, running on VM
- Embedded Workbench: Arm
- Embedded Workbench version: 9.1
- VSC Extension version: IAR Build v1.30.6, IAR C-SPY Debug v1.30.4

**Additional context:**
Add any other context about the problem here, e.g., log, projects, files etc. Logs can be obtained from the output view under the "IarServiceManager", "IAR Build", "IAR Config Generator" and "C/C++" sections.

mario-pi, 2024-07-26:
Hello,

I tried to reproduce this issue and, as far as I can see, I can only see correct results in autocompletion, see screenshots below.
[image]
[image]

In order to further diagnose the issue it would be useful to look at a basic example project, including the ewp file, and also know which exact version of EWARM (9.10, 9.20, etc...) you are running against.

Kind regards,
M

justric123, 2024-07-29:
Hi Mario, 
thanks for the quick reply. I was able to reproduce your example, it works fine for me as well. Furthermore I was able to narrow down the problem. Including C headers seems to be the cause of the unrelated autocompletion:
[image]
[image]

Without the C header include the class autocomplete is correct. But in IAR the autocompletion works fine even with the c header.
In VSC I still get the note:
"Note: IntelliSense is not fully configured. Use the 'Select IntelliSense Configuration...' command to finish configuration."

IAR ARM 9.32.2.57414
shared comp. 9.1.8.10480
IAR C/C++ Compiler for ARM
  9.32.2.340 (9.32.2.340)

I'm not sure which details you need from the ewp file:
...
    <option>
        <name>OGProductVersion</name>
        <state>5.10.0.159</state>
    </option>
    <option>
        <name>OGLastSavedByProductVersion</name>
        <state>9.32.2.57275</state>
    </option>
...
     <name>ICCARM</name>
      <archiveVersion>2</archiveVersion>
      <data>
          <version>37</version>
          <wantNonLocal>1</wantNonLocal>
          <debug>1</debug>
...
       <name>AARM</name>
        <archiveVersion>2</archiveVersion>
        <data>
            <version>11</version>
            <wantNonLocal>1</wantNonLocal>
              <debug>1</debug>
...

Thanks and regards,
Alex

mario-pi, 2024-07-30:
Thanks for taking the time to further diagnose this. We'll try to look into this as soon as possible, but I cannot promise any exact time frame at this point.

HampusAdolfsson, 2024-08-22:
It might be helpful to look at the logs for the `C/C++` extension. Change this setting to Information or Debug:
[image]
then restart your VS Code window. If you press Ctrl+Shift+U, and select `C/C++` in the dropdown on the right, is there anything there about why intellisense is not configured correctly?

It's possible that there is something in the header (or one of its inclusions) that is not being parsed correctly by the `C/C++` extension. Typically, this can happen with code that uses IAR language extensions. Does VS Code report any errors/squiggly lines for your headers?

justric123, 2024-08-22:
Thanks for the suggestions. It kind of found a workaround because I'm not able to really find the wrong setting. I deleted the User folder in AppData (C:\Users\Admin\AppData\Roaming\Code\User). After that it seems to work fine. If this won't be a long term solution I will have a look on it again.