iarsystems/iar-vsc-build #82: Error initiating IAR toolchain backend. Some functionality may be unavailable: Service registry exited prematurely, code: 1

Anc9, 2024-08-02:
**Describe the bug:**
This error occurs when I open the iar project on vscode, resulting in the files directory in iAR-build not being displayed. I have looked up some other bugs, but no similar problem has occurred, please help me confirm what is the problem?I tried to reinstall iar and vscode, but it still didn't work. The iar version is 9.30.1
The following is the output of IarServiceslauncher
[image]
[image]

**To Reproduce:**
Describe the steps to reproduce the problem in as much detail as possible.

**Expected behavior:**
A clear and concise description of what you expected to happen.

**Actual behavior:**
What actually happens.

**Environment:**

- OS: [Windows, linux]
- Embedded Workbench: [Arm, RX, RL78, AVR, RISC-V, ...]
- Embedded Workbench version:
- VSC Extension version:

**Additional context:**
Add any other context about the problem here, e.g., log, projects, files etc. Logs can be obtained from the output view under the "IarServiceManager", "IAR Build", "IAR Config Generator" and "C/C++" sections.

mario-pi, 2024-08-02:
Thanks for reporting this issue. There seems to be a problem with how the .json files are read from the service launcher utility.
Could you please provide more details about your system configuration (including the system OS and its version, language and locale settings), so that we can try to reproduce this?

Anc9, 2024-08-03:
Thank you for your reply. My system is win10 and the language is Chinese. I would like to add that every time I reinstall iar, the file view can be displayed normally when I open an iar project using vscode for the first time, but when I close vscode, I open the same iar project again. The above problems arise. Thanks for your reply and have a great weekend.
Below is my version of windows：（By the way, both my vscode and iar-build extensions are the latest versions）
[image]

Anc9, 2024-08-28:
Thanks for your reply, I think I found the problem, due to the company, after I open this json file, the computer I work on encrypts the json file, which causes the next time I open the vscode again, it looks like the iar expansion plugin can't read this file, so this problem occurs!

mario-pi, 2024-08-29:
Thanks for following up on this. As the issue is not directly related to the VSCode plugins, you should either report this to your IT department or to the IAR technical support, so that we can know more details about your configuration.

Closing.

closed 2024-08-29 by mario-pi