iarsystems/iar-vsc-build #84: Plugin can not find the project file location: ENOENT: no such file or directory, stat

dudamao, 2024-09-09:
**Describe the bug:**
Hey friends in IAR, I encountered a path recognition issue while using this plugin. Although a project file exists in the folder, the plugin indicates that it cannot find the file. I suspect a possible cause is that the plugin automatically adds spaces when recognizing the path.

[image]

file path: C:\main\dev\MCAL_Motor\iar\iar_v850\E3_ref_gateway_E3640

error:
Could not parse project file 'c:\main\dev\MCAL_Motor\iar\iar_v850\E3_ref_gateway_E3640\E3_ref_gateway_E3640.ewp ': ENOENT: no such file or directory, stat 'c:\main\dev\MCAL_Motor\iar\iar_v850\E3_ref_gateway_E3640\E3_ref_gateway_E3640.ewp '
[image]

**To Reproduce:**
using a VSCode Workspace include project folder can reproduce this issue.

**Environment:**

- OS: [Windows, linux]
- Embedded Workbench: Arm
- Embedded Workbench version: 8.50.6
- VSC Extension version: v1.30.6

HampusAdolfsson, 2024-09-13:
Hi,

Could you please check your workspace file (`E3_ref_gateway_E3640.eww`) to see whether there might be a space at the end of the project path, as in this example:
[image]

The extension reads the path directly from the .eww file, so a space at the end would cause the error you described. Note that the Embedded Workbench IDE is able to find the right file even with a space at the end of the path, which would explain why you're only seeing this error in VS Code.

dudamao, 2024-09-24:
Hi Hampus, Thank you very much for your explanation. This was exactly my issue, and after modifying the .eww file, my problem was solved. I hope that in future plugin updates, this suggestion can be added to the error message.

closed 2024-09-24 by dudamao