iarsystems/iar-vsc-build #85: Size of enum is shown wrongly in Intellisense

Shulcare, 2024-11-22:
**Describe the bug:**
When defining enums with values that can fit into a char or short, sizeof() of this enum typedef always shows 4 on mouser hover. This is also reflected in sizeof structs that use these enums which get much bigger than they should.
I've tried to find any Intellisense configuration to change this, but did not find anything.

**To Reproduce:**
Define an enum typedef with max value <= 0xFF. Write sizeof(<this type>) and hover on it.

**Expected behavior:**
Sizeof enum should match the actual size from the compiler, which in case of IAR is always the smallest possible type to fit all values. So it should show 1 or 2 for smaller enums.

**Actual behavior:**
Size of enum types is always 4, no matter how small they are.

**Environment:**
- OS: Windows 11
- Embedded Workbench: Arm
- Embedded Workbench version: 9.40.1
- VSC Extension version: 1.30.6

**Additional context:**
There are no flags set in our IAR project that would change the enum sizes from their default, and the ELF output definitely has the smaller enum sizes.

HampusAdolfsson, 2024-11-26:
Hi,

This extension relies on the `C/C++` extension to provide intellisense, which is mostly focused on clang, GCC and MSVC. What you are seeing is likely the size clang would use for the type. Their intellisense engine is closed-source, so unfortunately there isn't much we can do from our end. There is a ticket open at microsoft/vscode-cpptools#653 to provide better support for IAR compilers, but there hasn't been much activity on it.

Shulcare, 2024-11-26:
Hello,
Ok, then I guess this issue will stay for a while...  I would be fine with adding some compiler flag like -fshort-enums to force the small size, but unfortunately this leads to an aborted build because the IAR compiler does not know this. Adding __attribute__ ((__packed__)) to the enum declaration does work, but I don't want to change the source code.
Is there maybe any other workaround for this?