iarsystems/iar-vsc-build #87: Add the possibility to re-search workspaces/projects in the current workspace

axelnxp, 2025-02-19:
My workspace contains several IAR projects that are generated with a tool, this means I often need to re-generate my projects.
When I do this, the IAR Build extension doesn't find the newly generated project, I need to "Developer: Reload Window", which is not very convenient.
It would be greate to have a button or an action which allows to search for eww and ewp files and get an updated list of available projects.

HampusAdolfsson, 2025-02-26:
Hi!

This is something the extension is meant to handle automatically when it detects file changes in your VS Code workspace folder(s). Could you describe a bit closer what your use case is? It's possible you've found a bug or have a use case we haven't thought of.

The extension listens for new/deleted .eww files, and refreshes the list of workspaces when it detects one. Additionally, it listens for modifications to the currently selected .eww file, and refreshes the list of projects in the workspace when the .eww file is changed.

axelnxp, 2025-02-26:
Hi @HampusAdolfsson 

Ok, from what I see, the extension doesn't react immediately to these kind of changes, how does it work ? Is it a periodic check like every 5min it performs a file scan ? Or is it supposed to react immediately to any change ?

My workspace is quite big, it's a whole MCU SDK, so maybe the extension doesn't catch the change due to the number of files in the workspace ?

In my workspace, the IAR projects (.eww and associated .ewp) are generated with a tool. I often re-generate them, so the original file gets replaced by the new one, but the path is the same, the name of the files is the same etc.

If I generate a new project, meaning it doesn't exist in my workspace, the extension doesn't detect it, I need to reload the window and only then it detects it.

If I regenerate a project that is currently opened in the extension, I observed 2 different behaviors
* it doesn't do anything, I can just reload the project and build it correctly
* it assumes the project doesn't exist, the extension opens another project (the first one in its list I think) and if I try to re-open the previous project, it doesn't appear in the list of available projects

HampusAdolfsson, 2025-02-26:
Thanks for the clarification.

We use a file watcher API built into VS Code itself, which responds immediately to changes. In our tests, we've seen that this file watcher sometimes fails to detect changes when they happen too quickly after one another. Specifically, if you delete a file and recreate it shortly after (rather than just overwriting it), it sometimes only detects the first event (the deletion).

It would be very helpful to us if you could post the extension logs produced when the bug occurs. This would tell us what file events the extension sees, and whether the extension responds properly by reloading the workspace. You can find the logs in the `Output` window (ctrl+shift+u), by selecting "IAR Build" in the dropdown on the right. This contains project paths and configuration names, so please make sure this isn't sensitive information.

axelnxp, 2025-02-27:
@HampusAdolfsson I could get some logs when I re-generate the project.
Here is the scenario:
* The project already existed, and I had it open in the extension
* I re-generated the project (so delete + re-create), and I still had it open in the extension, no logs output
* Few minutes later, the extension closed the project, and I can't find it anymore in the project list, I got some logs:
```
[2/27/2025, 3:01:48 PM][Debug] Detected deleted file: /path/to/myproject.eww
[2/27/2025, 3:01:48 PM][Debug] Found 134 workspace(s) in the VS Code workspace
[2/27/2025, 3:01:48 PM][Debug] Selected workspace: $[project_name]
[2/27/2025, 3:01:48 PM][Debug] Detected deleted file: /path/to/myproject.ewp
[2/27/2025, 3:01:48 PM][Debug] Found 179 project(s) in the VS Code workspace
[2/27/2025, 3:01:48 PM][Debug] (bxarm (Arm)) Disposing of workspace: myproject
[2/27/2025, 3:01:48 PM][Debug] Loading thrift workspace: $[project_name]
[2/27/2025, 3:01:48 PM][Debug] Loaded workspace '$[project_name]'
```
So the weird thing here is that the extension detects the deletion several minutes after the project file was re-created.

HampusAdolfsson, 2025-03-04:
The file watcher has a few limitations:

* It does not detect changes for files that are not inside your VS Code workspace folders
* It does not follow symlinks

However, since you still *get* the events, I doubt either of these are the problem. I think there are two possibilities here:

* The large number of watched files causes some sort of resource shortage in the underlying file watcher or the OS. File watchers are quite resource-intensive and sometimes unreliable.
* The extension itself is stuck doing something else and doesn't have time to process the events for quite some time.

Could you take a look at VS Code's own file watcher logs, to see if it also receives the events late or if it's just the extension? You'll need to run `Developer: Set Log Level...` from the command palette, then select `File Watcher` and then `Trace`. Then, you can find the logs in the "File Watcher" category in the same Output window as before.

You should see something like this in response to file changes:
```
2025-03-04 11:02:02.717 [trace] [File Watcher ('parcel')] [DELETED] /path/to/project.ewp
2025-03-04 11:02:02.793 [trace] [File Watcher ('parcel')]  >> normalized [DELETED] /path/to/project.ewp
```