iarsystems/iar-vsc-build #89: Question: checking memory at specified location

MasaitisErikas, 2025-03-17:
Is there a way to search for exact memory location (like built-in tool for IAR) ?

jlonnberg, 2025-03-18:
@MasaitisErikas If I understand the question right, yes. The addition of listwindows in the lastest release added support for the Symbolic Memory Windows which allows you to search the memory for variables etc. The view can be shown by selecting it from the context menu in the debug view, see image below.

Use the toolbar to search for a specific memory location or variable name.

[image]

MasaitisErikas, 2025-03-18:
Search bar is not visible for me. Maybe I need to turn it on somewhere?

[image]

jlonnberg, 2025-03-18:
@MasaitisErikas Which OS, architecture and version of Embedded Workbench are you using?

MasaitisErikas, 2025-03-18:
OS: Windows 11
    Embedded Workbench: RL 78
    Embedded Workbench version: Embedded Workbench 9.1_2
    VSC Extension version: 1.30.7

jlonnberg, 2025-03-18:
@MasaitisErikas Seems to be a bug. We will try to fix it in the upcoming release