iarsystems/iar-vsc-build #90: Cannot use relative -I argument in options file (cwd option seems ignored)

LilMouse, 2025-03-24:
**Describe the bug:**
I have a project that has the extra C/C++ command line compile option `-f $PROJ_DIR$\options_file.`
In this options file I have `-I..\include_dir`

When building with iarbuild.exe from the right folder this works.
```
cd <folder with ewp file>
'C:\Program Files (x86)\IAR Systems\Embedded Workbench 8.0/common/bin/iarbuild.exe' <my_project_location>.ewp -make Debug -log info -parallel 4
```
But with the iar build extension it seems to run from a different directory.

It complains that it cannot find the file in this `..\include_dir`
```
Fatal Error[Pe1696]: cannot open source file "test.h"
            searched: "..\include_dir"
```
I have also tried setting the "cwd" in task.json

```
"options": {
"cwd": "D:/<where_my_folder_is_located>",
},
```

**To Reproduce:**
Describe the steps to reproduce the problem in as much detail as possible.

Create a project with an options file and set the extra commandline options in the ewp file
```
<option>
    <name>IExtraOptions</name>
    <state>-f $PROJ_DIR$\options_file</state>>
    <state>-D SETEMULATOR</state>
</option>
```

Set the task.json options.cwd to ${cwd}
```
"options": {
    "cwd": "${cwd}",
},
```
Build the project in vscode with iar build task.

**Expected behavior:**
A clear and concise description of what you expected to happen.

The build is run from the project folder/cwd and can find the relative include directory.
The command iarbuild should be run relative to the working directory/ewp file location.

**Actual behavior:**
What actually happens.

The build cannot find the directory/files inside `../include_dir`.
I am assuming the directory the command is run from is different than I expect so it cannot find `..\include_dir`

**Environment:**

- OS: Windows
- Embedded Workbench: Embedded Workbench 8.0 MSP430
- Embedded Workbench version: 8.0 MSP430
- VSC Extension version:  1.30.7

**Additional context:**
Add any other context about the problem here, e.g., log, projects, files etc. Logs can be obtained from the output view under the "IarServiceManager", "IAR Build", "IAR Config Generator" and "C/C++" sections.

HampusAdolfsson, 2025-03-25:
Hi, thanks for the detailed description!

When the extension runs iarbuild, it sets the working directory to VS Code's "workspace directory", i.e. the root of the directory you have opened in VS Code. If your .ewp file is in this directory you should get the behaviour you expect, but I suspect in your case the .ewp file is in some other directory and I can see why you might want to change the working directory then.

We can make sure the extension picks up the `options.cwd` field from tasks.json and sets the working directory for iarbuild accordingly (by default, I think we should stick to using the workspace directory as cwd so that we do not break existing tasks).

HampusAdolfsson, 2025-06-10:
We just released version `1.32.1` of the extension with a fix for this issue. You can now specify the working directory in your tasks.json file as you suggested:
```json
"options": {
    "cwd": "<working directory>",
},
```

closed 2025-06-10 by HampusAdolfsson

LilMouse, 2025-06-10:
Tested this and it works great. Thanks for the fix!